SJSURoboticsTeam/urc-central-2021 #21: [ DEMO ] Create plug-able terminal block patterns

kammce, 2020-06-20:


kammce, 2020-12-02:
@SJSURoboticsTeam/electrical do we actually already have a version of this?

QuanLe8191, 2020-12-02:
I dont think we currently have any. Are we creating a pattern for the Phoenix Connector?

kammce, 2020-12-03:
Yup, the ones that you want to design around look like this:

## Board mount parts

Link: https://www.aliexpress.com/item/33040399030.html?spm=a2g0s.9042311.0.0.27424c4dY8gU4t

[image]

##  Connector part

Link: https://www.aliexpress.com/item/33040591852.html?spm=a2g0s.9042311.0.0.27424c4dY8gU4t

[image]

The dimensions (and equations) can be found if you scroll down but I'm also adding the images here.

QuanLe8191, 2020-12-04:
Arent we have a terminal block 2.54mm footprint? This is a new one?

kammce, 2020-12-05:
Oh ok, then that looks fine then. May be a good idea to add the shape of header block on the component as well to prevent EEs from putting these too close to each other.