SJSURoboticsTeam/urc-central-2021 #47: Determine handover from Equipment Servicing Mission to Autonomy Mission (target time 10 minutes)

ntwong0, 2020-06-25:


ntwong0, 2020-06-25:
[image]

Previous mast

ntwong0, 2020-06-25:
Latest design as per @wilton-lee 
[image]

Other notes:
* It's small enough to fit horizontally inside the chassis so that could be a possibility if it would be better to have there instead of strapped on the side of the mast
* (camera's are not shown because I need to fix the orientation depending on which of the above choices we choose to go with)
* David also said something about buying duplicate mast/communication equipment so we would take off mast/arm entirely after that part of the competition and replace with the intel box with the duplicate mast/communication equipment, not sure what your opinion is on that
* so currently the 3 potential options I can think of are:
    1. box inside chassis
    2. box strapped on side of mast
    3. replace arm/mast with box+duplicate mast equipment

ntwong0, 2020-06-25:
Reasons against chassis option

1. wiring complexity, we'd still need to run the GPS receiver up to the top of the mast
2. camera swap complexity. use of the gimbal/camera/rpi turns out to be an issue in the past, where the video feed eats up the available bandwidth, so we are considering swapping it with USB camera for the autonomous nav mission (altho this could also be mitigated by reducing the framerate)

I actually don't see #2 as a big one because I could go either way with it, but i'll let that stand nonetheless. If the wiring complexity isn't a problem, and there are no complications with inserting the box into the chassis, then I have no complaints.

wilton-lee, 2020-07-02:
Going to proceed with having the box strapped to the side of the mast.

ntwong0, 2020-07-11:
@wilton-lee can you provide an estimate for the materials and access to the 3D model assets by next Saturday?

ntwong0, 2020-07-22:
Migrating comments from #53 

> From 7/18 @wilton-lee
>> so for the autonomy period, I am thinking of replacing the mast/arm assembly with the intel box with duplicated mast equipment needed during the autonomous period. Which of these mast equipment would we need to duplicate if we went this route?
>> 1. see3cam 10x zoom camera
>> 2. LIDAR-lite laser rangefinder
>> 3. Runcam Swift camera
>> 4. Tracker yaw & pitch motors
>> 5. Ethernet/Networking switch
>> 6. Ubiquiti Nano station rocket m900
>> 7. 900 Mhz 5 dBi Rubber duck antennas
>> 8. any other equipment I am missing?

> REF: URC Budget and Components
> * Items 1-4 not necessary.
>     * Replace with line item for `Logitech HD Pro Webcam C920`
> * Items 5-7 correspond to teleoperations & comms, we need everything rover side
>     * The spreadsheet is currently missing: `rover side wireless router`

Followed by

> @ntwong0
> So all the equipment needed for the autonomous period besides equipment specified for the intel box would be:
> 1. Logitech HD Pro Webcam C920 - $80
> 2. Rover side wireless station: $160
> 3. Rover side external antenna: $24
> 4. Rover side wireless router: ??
> Do you know the cost of the wireless router? and is the Logitech webcam also being used during tele-op or strictly for autonomous?

ntwong0, 2020-07-22:
@wilton-lee 

For items 1-3, yes I believe so. 
For item 4, the `rover side wireless router` should match with the one selected by @mast, although I don't know who has taken charge of that.
For item 1, this USB camera is specific for the autonomy mission, and is used for both tele-op mode and autonomous mode.

ntwong0, 2020-07-23:
@PushpalPatil @YMmirsky can you meet with @josh-github1 at some point to hash out decision making for mast? We need to know who will take on the responsibility of selecting the wireless router

wilton-lee, 2020-08-15:
@HaadiElahi

QuanLe8191, 2020-08-20:
I will look into the datasheet for them and change the autonomy block power diagram based on the new information

wilton-lee, 2020-08-20:
Mid week update:

- using 6 cut-to-size acrylic sheets instead of 3d printing
- currently designing mounting brackets 

To Do:

- send Quan dimensions for electronics boarad

QuanLe8191, 2020-08-23:
Have we decided on which router, antenna and wireless station we gonna use?

QuanLe8191, 2020-08-27:
https://www.amazon.com/dp/B07P5QDQ1B

wilton-lee, 2020-09-03:
[image]
to do:
- outtake fans on top panel
- vents on back panel

QuanLe8191, 2020-09-04:
Router: Raftpower RD-W009
Ubiquiti Nano station rocket m900
900 Mhz 5 dBi Rubber duck antennas

wilton-lee, 2020-09-10:
https://www.amazon.com/Pura-Computer-Air-Filter-Polyurethane/dp/B000FJU8BC/ref=sr_1_11?dchild=1&keywords=fine+dust+filter&qid=1599706308&s=electronics&sr=1-11