SJSURoboticsTeam/urc-central-2021 #48: Gyroscope/compass Linux/Jetson Nano driver

ntwong0, 2020-06-25:
progress
- [x] get raw data from MPU-9250
- [x] Decode compass data
- [ ] read data with ros-mpu9250-node
- [ ] Determine compass calibration procedure

useful link
- https://github.com/wolfeidau/ros-mpu9250-node

ref for i2c timing diagram
[image]

ntwong0, 2020-06-25:
No updates for this week, will check back on @BobMak next week

BobMak, 2020-06-27:
[image]
errors in making upm from the source. Need to resolve dependencies. so far tried apt-get install libbacnet libmodbus libopenzwave libopenzwave

ntwong0, 2020-07-11:
@BobMak can you attempt building `upm` from source by next Saturday? Stretch goal - draft of calibration procedure

ntwong0, 2020-07-21:
@BobMak @YMmirsky need clarification on next steps

ntwong0, 2020-07-23:
@BobMak still blocked on `upm` building, will meet with @YMmirsky to resolve

BobMak, 2020-07-23:
Able to build upm. upm requires a specific nodejs version to build optional utilities. It is possible to build upm without node by running `cmake <upm_directory> DBUILDSWIGNODE=OFF` 
[Original solution](https://github.com/eclipse/upm/issues/225)

PushpalPatil, 2020-07-30:
@BobMak Note: saturday

BobMak, 2020-08-01:
Rebuilding upm with raw=TRUE results in the same error

BobMak, 2020-08-08:
[Core issue](https://github.com/eclipse/mraa/issues/997). Replacing mraa i2c read/write calls in upm.

BobMak, 2020-08-20:
I've replaced the mraa i2c calls with linux i2c code in the upm as a temporary workaround. I can see the accelerometer data, but the orientation data looks broken.

PushpalPatil, 2020-08-27:
maybe use magnet method

BobMak, 2020-09-08:
Tried enabling magnetometer by setting the `enableAk8975` flag to true. Still getting garbage data, no noticeable changes when the magnet is put closer.

PushpalPatil, 2020-09-09:
can you post an example or vid recording of the garbage data? also could you push your code to the repo so that we can see the code?

BobMak, 2020-09-12:
[garbage data and magnet test](https://youtu.be/cJh2a9UlK9U)

BobMak, 2020-10-04:
Failed to utilize the ros-mpu9250-node module on Jetson Nano. The low-level dependency of the node doesn't support Jetson nano yet. Replacing the dependency failed. Working on my own magnetometer node implementation. Already able to initialize and read from the magnetometer.

BobMak, 2020-10-22:
working on a python mpu9250 orientation node implementation.

BobMak, 2021-01-11:
mpu orientation data looks consistent and is reacting to a magnet.
now trying to figure out how to do self-test and how to calibrate the sensor.

vizualised pose data by using [Nelson's rviz notes](https://github.com/SJSURoboticsTeam/urc-autonomy-subsystem-2021/tree/mpu9250_viz)
there was one other thing I needed to run to see the live data visualisation:
```rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0 0.0 0.0 map imu 100```

BobMak, 2021-01-14:
[mpu9250 compass heading visualization](https://www.youtube.com/watch?v=zxfoSmDnpBw)