SJSURoboticsTeam/urc-central-2021 #53: Meet with EE to discuss electrical for ACM

ntwong0, 2020-06-25:
[NVIDIA Jetson Nano Developer Kit User Guide](https://drive.google.com/open?id=1Jn9KSZIr1M0lHHIbM-EjLl1gRLlBBSg_)

- page 7: accepts 5V 4A, 4.4A max continuous current. Power jack dimensions 2.1x5.5x9.5mm plug with positive polarity
- page 9: requires 5V 2A

[NVIDIA Jetson TX1/TX2 Developer Kit Carrier Board Specification Sheet](https://drive.google.com/open?id=1xi71eoMjb_VQApkxA47YtmS9uMxWkvNf)

- page 32: DC Power Jack specification. Jack make/model Singatron Enterprise 2DC-213-B51. Mating plug is indicated as Singatron Enterprise 2DP-313-B01
- page 38, table 35: VDD_IN/VDD_MUX, main power input from DC adapter. V range 5.5-19.6 V, max current ~4000 mA
- note, the supplied power adapter (shipped with the developer kit) is rated to 90 W

[Intel RealSense D435i Datasheet](https://drive.google.com/open?id=1xBYPFxBK0hiu7kZYjW-iQkWVXLuENr8A)

- Note: USB port is USB Type C
- page 49, table 3-39: VCC 5.25 V max, ICC 700 mA max (so theoretical max IV = 3.7 W)
- page 62, tables 4-10 and 4-11: D430-series max measured power 2.85 W

[Intel RealSense T265 Datasheet](https://drive.google.com/open?id=16rASY6cpy1XLIQVWMF5-r6P7jwDEYa7q)

- Note: USB port is USB 3.0 Micro B
- page 11, table 3-5: VCC 5.25 V max, ICC 300 mA max, so theoretical max IV = 1.575 W
- no measured power data provided

ntwong0, 2020-06-25:
## Importing from Trello

### Jun 8 at 7:49 PM
Waiting on Smruthi RE: 6S battery, need to confirm how much power the current battery has

### Jun 4 at 6:09 PM
quan_le1 working on power distribution board

### May 30 at 11:46 AM
quan_le1 as1 indicated by wiltonlee the max footprint for the power distro board for ACM Rev2 is 170mm x 100mm

He also indicated

> keep in mind any wires that might stick out from the sides when connected to the board

### May 30 at 11:46 AM
Meeting with Quan and Smruthi on 5/29, discussed rough estimates for ACM battery req's on ASDD slide 47

* quan_le1 tasked to explore power distribution for Jetson TX2 and Jetson Nano
* ntwong0 task to provide electrical and connector info, along with datasheet
* wiltonlee can you let us know the footprint that quan_le1 has available for power distribution, based on ACM Rev2 from ASDD slide 14?

QuanLe8191, 2020-07-01:
Considering using a simple DC-DC buck with a low voltage cut-off
Low voltage cutoff: https://www.amazon.com/Actobotics-Low-Voltage-Cutoff-Switch/dp/B0786RK98Y

ntwong0, 2020-07-02:
Crosslinking ACM design from #47 to here, specifically [this revision](https://github.com/SJSURoboticsTeam/urc-central-2021/issues/47#issuecomment-649195546)

ntwong0, 2020-07-11:
@QuanLe8191 can you send me a mockup of the power distribution unit by next week?
* LiPo input
* 5V output to Jetson Nano
* 19V output to Jetson TX2

QuanLe8191, 2020-07-19:
Add block diagram for the autonomy box: https://app.lucidchart.com/invitations/accept/1ac9491f-226d-415f-9765-6f33a440e61f

ntwong0, 2020-07-21:
@wilton-lee can you attach your updates here

ntwong0, 2020-07-21:
From 7/18 @wilton-lee 
> so for the autonomy period, I am thinking of replacing the mast/arm assembly with the intel box with duplicated mast equipment needed during the autonomous period. Which of these mast equipment would we need to duplicate if we went this route?
1. see3cam 10x zoom camera
2. LIDAR-lite laser rangefinder
3. Runcam Swift camera
4. Tracker yaw & pitch motors
5. Ethernet/Networking switch
6. Ubiquiti Nano station rocket m900
7. 900 Mhz 5 dBi Rubber duck antennas
8. any other equipment I am missing?

REF: [URC Budget and Components](https://docs.google.com/spreadsheets/d/1pIhI2VJR_fADTYlfWnPg5Y6lvLI7TMzRsGy_LCxZKhc/edit#gid=1704595482)
* Items 1-4 not necessary.
    * Replace with line item for `Logitech HD Pro Webcam C920`
* Items 5-7 correspond to teleoperations & comms, we need everything rover side
    * The spreadsheet is currently missing: `rover side wireless router`

ntwong0, 2020-07-21:
@QuanLe8191 confirming mock up for power distro ready by Saturday 7/26

wilton-lee, 2020-07-21:
@ntwong0 

So all the equipment needed for the autonomous period besides equipment specified for the intel box would be:
1. Logitech HD Pro Webcam C920 - $80
2. Rover side wireless station: $160
3. Rover side external antenna: $24
4. Rover side wireless router: ??

Do you know the cost of the wireless router? and is the Logitech webcam also being used during tele-op or strictly for autonomous?

ntwong0, 2020-07-22:
@wilton-lee I just realized I've directed you to the wrong issue, so I'm going to redirect you to #47

ntwong0, 2020-07-23:
Updated [autonomy box](https://app.lucidchart.com/documents/edit/71805c18-4920-4c87-942a-6f492036afd2/0_0?shared=true) diagram, power and data to RealSense cameras provided by respective Jetson compute modules

ntwong0, 2020-07-23:
@QuanLe8191 please confirm make and model for buck converters and LiPo battery

ntwong0, 2020-07-23:
@QuanLe8191 for autonomy-mast, let's keep the same power topology as arm-mast w.r.t the communications equipment:
* power sources from LiPos in chassis
* connection from chassis LiPos to power distro board on mast
* power distro board supplies power to communications equipment

QuanLe8191, 2021-03-18:
I stumble upon this [buck converter](https://www.pololu.com/product/2851) the other day and I think it will be a great fit for the jetson nano