SJSURoboticsTeam/urc-central-2021 #67: Figure out donation situation

davidbdias96, 2020-06-27:


kammce, 2020-07-23:
The website for making donations changed internally, but I have access now.

I figured out the matching situation with Google. The amount that a Google matches employees can be found here: https://doublethedonation.com/matching-gifts/google-inc. 

I should be able to donate when y'all have access and control over the student club account. There is no reason to donate if you guys cannot get access to the funds in the first place.

kammce, 2020-07-23:
For reference at the time of writing this, this is the amount matchable:

[image]

closed 2020-07-23 by kammce