SJSURoboticsTeam/urc-central-2021 #73: determine the strength of drive motor needed

davidbdias96, 2020-07-08:
Figure out if we can buy a cheaper version of the drive motor by using gearing to reach the required output torque.

kammce, 2020-07-08:
We discussed in a previous meeting the potential to gear down the drive motor/servo using a timing belt. This had the benefit of allowing us to choose a smaller, lighter, and cheaper motor. But! when I looked around, I couldn't find any motors that were cheaper, lighter or smaller that also had the servo control we wanted. They were either a lot heavier, or just plainly heavier, or much more expensive. I think the motors I found are pretty much the best we got. But if we did want to slow down the speed of the motors and get insane amounts of torque out of them we can always gear down the motors. But the servo control means we can maintain a slow speed and high torque without needing an external system to help drive higher torque. Basically, gearing down just help with overall handling, but can definitely help with increasing our absolute maximum torque.
I'll continue to look though. The offset design may actually be simpler, but always keep assembly and number of components in mind. Always minimize the BOM line count as much as possible. Always push to make assembly easier and straight forward.

kammce, 2020-07-08:
I'd like to make the following assumptions:

1. Drive Motors: 3
2. Total Weight During Extreme Retrieval:  40kg

Why just for extreme retrieval? Because that is the only part of the competition where our rover is going to have to go through rough terrain and deal with inclines. At mostly level ground, the torque required to move is very small. In this case, we don't have to worry about the weight of science or intelligent systems. Just the drive systems and arm.

kammce, 2020-07-11:
With the following assumptions we can throw this data into this online calculator. Been using it for years now.

## Calculator Website

https://www.robotshop.com/community/blog/show/drive-motor-sizing-tool

## Wheel/Tire Used Previously

[image]

Link: https://www.ebay.com/itm/2pcs-13x5-00-6-Tyre-Tire-With-Tube-For-Mini-Pocket-Bike-ATV-QUAD-Gokart-Scooter/254292270541?hash=item3b35001dcd:g:HvIAAOSw0h1dJf1y

### Specifications: 

1. Inner tube
2. 13 inch diameter (7.5" radius)
3. 4" width

## Inputs to calculator

* Total Mass: **40kg**
* Number of wheels: **3**
* Radius of drive wheel: **7.5 in**
* Robot Velocity: **<ignore>** (does not contribute to torque requirement)
* Maximum Incline: **40 deg** (Anything more steep and we should probably try to avoid it rather than climb it)
* Desired Acceleration: **0.05 m/s^2** (this needs to be non-zero. If this is zero, then the torque we get as a result will be our stall torque. So choosing some small yet reasonable value is appropriate here to ensure we have the torque to keep moving)
* Desired operating time **<ignored>**
* Efficiency: **100%** (Set this value to 100% to ignore it. This particular field is confusing and not helpful. The torque output given by the calculator is based on the physics of the parameters given and we just need the minimum required given our inputs above.)

## Output:

Now read the torque value and that is what we need for the drive motor for the rover.

kammce, 2020-07-11:
This servo motor has more than enough torque for more than what is required for the rover:

## [RMD-X8 brushless dc servo joint motor](https://www.aliexpress.com/item/10000004036730.html)

[image]

[image]

kammce, 2020-07-25:
!> With the following assumptions we can throw this data into this online calculator. Been using it for years now.
> 
> ### Specifications:
> 1. Inner tube
> 2. 13 inch diameter ~(7.5" radius)~ (6.5" radius)
> 3. 4" width

# CORRECTION !! Torque required is **13.991** Nm

The used radius is incorrect. Should be 6.5in.

closed 2020-08-01 by kammce