SJSURoboticsTeam/urc-central-2021 #74: Email AS about accessing funds

davidbdias96, 2020-07-08:
Find out if we can remotely access our account in order to deposit and withdraw funds

davidbdias96, 2020-07-22:
Documents signed and sent to AS, Waiting on response.

kammce, 2020-07-23:
Can we document the process for remote access of funds? So that the process is transparent to the team. Make sure to not include sensitive information like the club account number etc.

davidbdias96, 2020-07-30:
Cash deposits: Make an in-person appointment with Vivian (Wednesday or Thursday)
Checks and money orders can be mailed in (Don't endorse checks)
There is only someone in the office once a week at the moment so any mailed in checks will have about a 2-week wait before funds are available.

Reimbursements/Requisitions: Need name and address of the person the money is being sent to.

Vivian.Nguyen@sjsu.edu  Office Manager (408)786-8488 cell phone

Helen.Nguyen@sjsu.edu  The Accountant

kammce, 2020-07-30:
Yikes. But this clears the way for funding. I'll see what I can do.

closed 2020-08-08 by davidbdias96