SJSURoboticsTeam/urc-central-2021 #79: Emergency Button

QuanLe8191, 2020-07-18:
Designing emergency button for the drive system

maximvoven, 2020-12-02:
It would be beneficial if emergency button affected both drive and arm simultaneously, but this is likely not feasible.

kammce, 2020-12-03:
# Why not 2 E stops?

We could also use two E-stop buttons, one for drive and one for arm. At least for the first iteration of the rover.

# Additional Idea: Galvanic isolated switch

Potentially after the first iteration of the rover we can do something fancy like, use a electromagnet that is powered by drive system that is cut off when power is eliminated. The other Arm Board has a hall sensor that is looking for that magnetic field and when it dies out, it disconnects the power to the arm. An alternative would be a light sensor. Or we could come up with some other spooky action at a distance way of cutting off the power Arm. 

NOTE: That my idea above does not consider how we make this work with a rotating arm since the part of the Arm that is electrically active is above the rotunda and moving. So we would need some ring of electromagnets or some LED array of lights to make this work. Ring of glowing LEDS sound pretty bad ass me NGL.

# Additional ADDITIONAL Idea: Bring back the slip ring

Utilize the same, or potentially modified, version of the slip ring used for drive steering to simply power to the E-stop circuit and when drive's power goes down, so does the E-stop circuit. Still something that should be considered for a V-2 of the rover. But I think that having a switch for Arm is critical so that we can turn both sides on and off.

kammce, 2020-12-03:
Adding mechanical label so they know that this is something they too have to prepare for. Controls is not required though as this system should knock out power without the need for controls to be aware as controls may be compromised at this point.