SJSURoboticsTeam/urc-central-2021 #86: Find belt pulley/sprocket to attach to bearing

HaadiElahi, 2020-07-18:
Modify bearing design to incorporate belt interface.

closed 2020-08-15 by HaadiElahi

kammce, 2020-08-20:
@HaadiElahi is this actually done or not? Also, can you put your findings here so that they are easier to access.

HaadiElahi, 2020-08-20:
@kammce I found [a pulley](https://www.mcmaster.com/57105K28/) that I attached to the slew bearing in CAD, but when we switched to the new motor I closed the issue.

kammce, 2020-08-20:
This still looks very much relevant so a better name for this issue should be something that includes the motor so that we know that it's closed that it isn't for any arbitrary belts that's on the Rover.

So I think changing the name of this issue and keep your closes when idea. Or we can open it back up and change his name to be relevant for the new motor.