SJSURoboticsTeam/urc-central-2021 #93: Materials for new pulley drive system

wilton-lee, 2020-07-25:
1. 1045 Carbon Steel Keyed Rotary Shaft Fully Keyed, 1/2" Diameter, 9" Long (https://www.mcmaster.com/1497K9/) - $19
2. Quick Detachable Bushing, QT Series, 1/2" Bore Dia., 1.250" Length (https://www.grainger.com/product/TB-WOOD-S-Quick-Detachable-Bushing-5UHZ6?) - $14 x 2?
3. Flanged Ball Bearing ID 1/2" x OD 1-3/8" (https://www.amazon.com/XiKe-Bearing-Wheelbarrows-Suitable-Replacement/dp/B01J67N29I/ref=sr_1_6?dchild=1&keywords=flanged+ball+bearings+.5in&qid=1595640159&sr=8-6) - $2 x 2
4. https://www.mcmaster.com/3389N12/ - $13 x 4 OR Clamping Two-Piece Shaft Collar for 1/2" Diameter (https://www.mcmaster.com/6436K14/) - $4.5 x 4???

wilton-lee, 2020-08-01:
Motor: RMD-X7 
- https://www.aliexpress.com/item/10000397235320.html?spm=2114.12010612.8148356.3.66fd22a06keXEs

Pulleys: 
- https://www.aliexpress.com/item/4000768114918.html?spm=2114.12010612.8148356.11.3b70e29dVvoN7N
- https://www.aliexpress.com/item/4000499698920.html?spm=2114.12010612.8148356.13.3b70e29dVvoN7N

Belt:
- https://www.aliexpress.com/item/32791799628.html

kammce, 2020-08-02:
@wilton-lee did you verify that the keyway, pulley width and everything else match? Because if you copied this from my presentation, I did not do all of the due diligence to verify that everything works. I just needed the materials to calculate approximate cost and weight.

wilton-lee, 2020-08-02:
I was using this list more to organize the links of potential parts to
choose from so I have quicker access. I will add a note to my main post to
clarify this.

wilton-lee, 2020-08-05:
Looking for 12mm keyed shaft ~200mm long OR 14mm flanged ball bearing OR compatible shaft+flanged bearing that can support loads

davidbdias96, 2020-11-06:
ready for purchase

closed 2020-11-06 by davidbdias96