SJSURoboticsTeam/urc-central-2021 #101: Design for quick detach rotunda

davidbdias96, 2020-08-03:


abentov, 2020-08-03:
[Trial Plate.zip](https://github.com/SJSURoboticsTeam/urc-central-2021/files/5019216/Trial.Plate.zip)

kammce, 2020-08-04:
What is in the zip file? Couldn't you link to our grab cad instead? @abentov 
Also, @abentov add your name to the github account so we know who you are.

davidbdias96, 2020-08-04:
Ohh yeah there was a bit of a miscommunication, I thought he was going to add some sketches so I told him to put them here but then they were actual part files so I had him put them in grab cad.

kammce, 2020-08-06:
Yeah, but who are they though?

davidbdias96, 2020-08-06:
Avi is the one working on it

kammce, 2020-08-06:
@abentov update your name on github.

davidbdias96, 2020-11-15:
completed

kammce, 2020-11-17:
Can some information regarding the detach rotunda be put here so that people looking back understand what happened? Like a link to the grabcad where the files reside would be great. Otherwise, it looks like this was completed after doing nothing.