SJSURoboticsTeam/urc-central-2021 #105: EE trial project

QuanLe8191, 2020-08-15:
Introduction
Candidates for the SJSU Robotics EE division should have a strong focus on schematic capture, and PCB layouts as well as the ability to extract information from a datasheet. This is used to test the student’s understanding of basic electrical components that we will be using for the rover throughout the course of the year. In addition to the EE skills 
required, you will be required to have basic knowledge of interfacing the electrical components with a microprocessor. 

This trial project will be evaluated by completion and by adherence to project 
rules/guidelines. 

Project Guidelines
Goal:
In this trial project, the DRV 8801 by Texas Instrument will be used to interfacing between a microcontroller of choice to a motor that required 24V – 3A

DRV 8801 is a common use motor driver in the SJSU robotic to control wheels and function of the drive system and arm system. 

Rules:
•	After reading through the datasheet of DRV 8801, design a circuit schematic that allow the microcontroller to communicate with the required motor. 
•	The designed circuit must be able to acts as an add on board (break out board) and show the ability to be tested independently.
•	A PCB schematic is also required and like the circuit schematic should be designed using EASYEDA.
•	You must work alone on this project. Working in teams is strictly forbidden.
•	You will have full flexibility on the project in term of micro-controller and circuit design
•	The designed circuit must follow the Robotics SJSU EE standard.

Notes:
Please be sure to understand the operation of your circuit, as many sizing calculators and tutorials are available online. You will be asked questions during your presentation, so please be prepared.
Any other questions or concerns may be redirected to the electrical lead, Quan Le through email: quan.p.le@sjsu.edu or through Slack.

Project Outcomes
•	The circuit follows the standard in the Robotics SJSU EE standard.
•	Circuit will be evaluated based on the completion of the circuit as well as your experience
•	Demonstrate understanding of the DRV 8801 datasheet
•	Understand the choice of passive component
•	DRV 8801 datasheet: 
•	Additional resource to help you on the project:

QuanLe8191, 2020-08-15:
[
[trial project 2020.docx](https://github.com/SJSURoboticsTeam/urc-central-2021/files/5079364/trial.project.2020.docx)
](url)

kammce, 2020-12-02:
Closing as this is no longer needed.

closed 2020-12-02 by kammce