SJSURoboticsTeam/urc-central-2021 #109: Battery Gauge Meter

QuanLe8191, 2020-08-21:


QuanLe8191, 2020-08-21:
Deciding between voltage and diode dropper:
       Voltage divider: lost of the accuracy of 28-30%

Right now go with the diode dropper that is stated in the doc
Schematic will have both so we can go through and see both of the option

nathaniel-nac, 2020-08-23:
Yes of course.

Battery Management Circuit Schematic: https://easyeda.com/editor#id=2b864c16d7eb4353bfe6344c9ca3e8e3

QuanLe8191, 2020-12-02:
I dont think we currently have any. Are we creating a pattern for the Phoenix Connector?

kammce, 2020-12-03:
> Actually now that I think about it again, would it actually be a good idea to add a BMS, just to ensure that all cells do discharge at somewhat the same rate?

Just so you know, very few BMS system controls the discharge of each cell during the discharge phase of a battery. Meaning when you power something using the BMS, the BMS's job is usually to turn off power in over current situations or in over temp situations. Fancier BMS do a bunch more but these are the basics. BMS DO handle discharging cells during the CHARGING of the pack to make sure that other cells do not get overcharged. So the cell balancing only happens when you are charging up the battery pack.