SJSURoboticsTeam/urc-central-2021 #123: Setup Dev2

naterpotatoers, 2020-10-03:
Had trouble setting up on windows.

ulysses10, 2020-10-09:
I ran into the same issue during setup on Mac (I installed the legacy VCP driver and tried the latest as well). I think it might be my connection speed since I get the same issue on Windows with a wired connection. I'm going to install Linux on a spare laptop and see how it goes. 
[image]

kammce, 2020-10-09:
Can you send the full log dump? just copy and paste it into this area here.

ulysses10, 2020-10-09:
Last login: Fri Oct  9 12:21:36 on ttys000
mactr0llii@Ulyssess-MacBook-Pro ~ % cd /Users/mactr0llii/Downloads/SJSU-Dev2-master
mactr0llii@Ulyssess-MacBook-Pro SJSU-Dev2-master % ./setup
./setup: line 30: /Users/mactr0llii/Downloads/SJSU-Dev2-master/tools/common.sh: No such file or directory
 ──────────────────────────────────────────────────┐
              Detecting your platform               
└────────────────────────────────────────────────── 
 ──────────────────────────────────────────────────┐
  List of software installed globally:              
   1) Python3                                       
   2) pip (Python3 package manager)                 
   3) Curl                                          
   4) Git                                           
   5) Gcovr                                         
   6) XCode CLI tools (OSX Only)                    
└────────────────────────────────────────────────── 
 ──────────────────────────────────────────────────┐
            Acquiring sudo privileges               
└────────────────────────────────────────────────── 
Password:

 ──────────────────────────────────────────────────┐

    Starting SJSU-Dev2 Environment Setup Script     

└────────────────────────────────────────────────── 
Operating System: Mac OSX Darvin
 ──────────────────────────────────────────────────┐

           Installing Mac OSX Prerequisites         

└────────────────────────────────────────────────── 
 ───────────────────────────────────────────────────┐
               Installing Homebrew                   
└─────────────────────────────────────────────────── 
./setup: line 44: [: -ne: unary operator expected
✔ - success
 ───────────────────────────────────────────────────┐
          Prompt Git installer by invoking it        
└─────────────────────────────────────────────────── 
git version 2.21.0 (Apple Git-122)
✔ - success
 ───────────────────────────────────────────────────┐
               Brew Install Python@3                 
└─────────────────────────────────────────────────── 
./setup: line 133: brew: command not found
 ───────────────────────────────────────────────────┐
                Brew Installing gcc                  
└─────────────────────────────────────────────────── 
./setup: line 137: brew: command not found
./setup: line 138: brew: command not found
./setup: line 140: 0 + : syntax error: operand expected (error token is " ")
 ─────────────────────────────────────────────────────────┐
                 Prequisite Install Failed.                
  Make sure:                                               
     - You are connected to the internet                   

└───────────────────────────────────────────────────────── 
mactr0llii@Ulyssess-MacBook-Pro SJSU-Dev2-master %

kammce, 2020-10-09:
What version of MacOSX are you using?

ulysses10, 2020-10-09:
Version 10.15.7 (19H2)
2018 Macbook

kammce, 2020-10-09:
What is the name of the OS? Catalina?

ulysses10, 2020-10-09:
Yes, Catalina

kammce, 2020-10-10:
That's weird. Should be working flawlessly, although the build for mac seems to be breaking all of a sudden but with a different setup issue thogh

ulysses10, 2020-10-10:
Setup on Linux :+1:

closed 2020-10-10 by ulysses10