SJSURoboticsTeam/urc-central-2021 #136: Drive Modes using Arduino, servos, and potentiometer

naterpotatoers, 2020-11-12:
Use the Arduino platform:

- Drive mode - Ulysses

- Translation/Crawl mode - Andrew

- Spin mode - Nate

Basically get a working drive mode prototype where you have servos that respond to potentiometer feedback. Where the servos mimic the wheels of the rover and the potentiometer acts as an angle input. Spin doesn't use angle input so  Use the rover [schema slides ](https://docs.google.com/presentation/d/1zS4pEHbo1Hao9_uDLm0cI-8jVlIfxareVtNrHyNaRSU/edit#slide=id.ga170ec689b_3_0) as a reference

naterpotatoers, 2020-12-03:
Spin Mode [video](https://www.youtube.com/watch?v=cqJhFKK6vtE&feature=youtu.be)

naterpotatoers, 2020-12-03:
All done, need to be done in SJTwo code

kammce, 2020-12-03:
We should post the arduino code and video for all of the modes here as a reference.

AndrewDEmerson, 2020-12-04:
Translation Mode [Video](https://www.youtube.com/watch?v=gwmaRFW4RXc)
Code also includes code to change which side of the rover is considered the front.
```C++
#include <Servo.h>

Servo wheel_1;
Servo wheel_2;
Servo wheel_3;

class Steering
{
private:
  // The angle that two legs of the robot makes
  const int servo_angular_offset = 60;
  const int servo_max_rotation = 180;
  char drive_mode;   // 'D' = drive mode; 'T' translation mode; 'S' = spin mode
  uint8_t front_dir; // number 1-3 that represents the side with the front 2 wheels
  Servo *wheels[3];
  uint8_t pins[3] = {8, 9, 10};
  Servo *back, *left, *right;
  int steer_angle, steer_speed;

  // Drive under translation (crab) mode logic.
  // Wheels will remain parallel to each other,
  // allowing the rover to drive straight in any direction.
  void mode_translation()
  {
    // NOTE: This code is for hobby servos, with a limited range of rotation
    // The actual motors have a much greater max rotation,
    // and should avoid having to flip their direction.
    Serial.println("Steering");
    if (steer_angle >= servo_angular_offset && steer_angle <= servo_max_rotation - servo_angular_offset)
    {
      back->write(steer_angle);
      left->write(steer_angle - servo_angular_offset);
      right->write(steer_angle + servo_angular_offset);
    }
    else if (steer_angle < servo_angular_offset)
    {
      back->write(steer_angle);
      left->write(steer_angle - servo_angular_offset + servo_max_rotation);
      right->write(steer_angle + servo_angular_offset);
    }
    else
    {
      back->write(steer_angle);
      left->write(steer_angle - servo_angular_offset);
      right->write(steer_angle + servo_angular_offset - servo_max_rotation);
    }
  }

public:
  // Attach the steering motors to the driver.
  // Motors should be assigned in a clockwise rotation,
  // where Servo A is the front-Right motor of side 1.
  Steering(Servo *A, Servo *B, Servo *C)
  {
    wheels[0] = A;
    wheels[1] = B;
    wheels[2] = C;
  }

  Steering(Servo *A, Servo *B, Servo *C, uint8_t steer_pins[3])
  {
    for (int i = 0; i < 3; i++)
    {
      pins[i] = steer_pins[i];
    }
    wheels[0] = A;
    wheels[1] = B;
    wheels[2] = C;
  }

  //Attach the Servos to the correct GPIO pins.
  void enable()
  {
    for (int i = 0; i < 3; i++)
    {
      wheels[i]->attach(pins[i]);
    }
  }

  // Update the steering system with new Drive-Mode, Angle, or speed.
  void updateSteer(char mode, int angle, int speed)
  {
    steer_angle = angle;
    steer_speed = speed;
    if (mode != drive_mode)
    {
      //Add Logic to get called when changing between drive modes
      drive_mode = mode;
    }
    switch (drive_mode)
    {
    case 'T':
      mode_translation();
      break;
    default:
      //throw drive_mode;
      break;
    }
  }

  //Change which side of the rover is considered the 'front' part.
  void changeFront(uint8_t front)
  {
    back = wheels[front % 3];
    left = wheels[(front + 1) % 3];
    right = wheels[(front - 1) % 3];
  }
};

Steering steer(&wheel_1, &wheel_2, &wheel_3);

void setup()
{
  Serial.begin(9600);
  steer.changeFront(2);
  steer.enable();
}

void loop()
{
  int pot = analogRead(2);
  float target_angle = map(pot, 0, 1023, 0, 180);
  Serial.println(target_angle);
  Serial.println(mode);
  steer.updateSteer('T', target_angle, 1.0f);
  delay(10);
}
```

ulysses10, 2021-01-11:
drive mode [video ](https://www.youtube.com/watch?v=1jEruVoMIkI&ab_channel=UlyssesVillegas)

Video shows 1 rear servo and 1 front servo, but code is for 2 front servos and 1 rear servo ( complete setup )
```C++
#include <Servo.h>

Servo servo1; // Rear servo
Servo servo2; // Front Left servo
Servo servo3; // Front Right servo

int potentiometer = A0; // Potentiometer Pin #
int potValue, potValue2; // Seperate values for front and rear servos

void setup() {
  servo1.attach(6);
  servo2.attach(7);
  servo3.attach(8);
}

//**Drive Mode Steering
void loop() {
  // Rear servo adjustment (mirrors front servos)
  potValue = map (analogRead (potentiometer), 0, 1023, 45, 135);
  servo1.write(potValue);

  // Front servos adjustment
  potValue2 = map (analogRead (potentiometer), 0, 1023, 135, 45);
  servo2.write(potValue2);
  servo3.write(potValue2);
  delay(15);
}
```

closed 2021-01-11 by ulysses10