SJSURoboticsTeam/urc-central-2021 #139: Psuedo Drive Firmware Schema

naterpotatoers, 2020-11-19:
As discussed in Thursday (11/19/2020) meeting, it would be nice to have a clearer idea of what would happen behind the scenes on the MCU side when mission control sends over their data. Basically, draft a pseudo function implementation for a drive mode that describes how to handle the data. 

Just so everyone is on the same page, here's a rough control loop that will decide which function mode to call.
```
while (true) 
{
  { speed, angle, mode } = getMissionControlCommands() // Isn't valid C/C++ code but assume we get the inputs here
  if (currentMode != mode)
    handleNewMode() // Might be a new story
  if (currentMode == drive)
    handleDriveMode(speed, angle) // Assigned to Ulysses
  if (currentMode == translation)
    handleTranslationMode(speed, angle) // Assigned to Andrew
  if (currentMode == spin)
    handleSpinMode(speed, angle) // Assigned to Nate
}
```

Add your pseudo implementations [here](https://docs.google.com/presentation/d/1zS4pEHbo1Hao9_uDLm0cI-8jVlIfxareVtNrHyNaRSU/edit#slide=id.ga170ec689b_3_0).

- [ ] handleDriveMode - Ulysses

- [ ] handleTranslationMode - Andrew

- [ ] handleSpinMode - Nate

naterpotatoers, 2020-12-03:
All done

kammce, 2020-12-03:
You say you are done, but where is the proof?

naterpotatoers, 2020-12-04:
Schema is implemented in [these](https://docs.google.com/presentation/d/1zS4pEHbo1Hao9_uDLm0cI-8jVlIfxareVtNrHyNaRSU/edit#slide=id.ga2c7c7c9bb_0_0) slides currently. Look at [this](https://github.com/SJSURoboticsTeam/urc-central-2021/issues/136) issue for implementation demos.

closed 2020-12-04 by naterpotatoers