SJSURoboticsTeam/urc-central-2021 #140: Shoulder Gear Ratio/Design

wilton-lee, 2020-12-01:
**SHOULDER:**
Required Torque: 65 N-m
Desired Gear Ratio: 10:1

_**Stage 1 Gears:**_

- [Gear 1- Motor Gear:](https://www.aliexpress.com/item/4000896303975.html?spm=2114.12010612.8148356.5.272b39fdOe3OUN) 1.5M-15T w/ 10mm bore x 4mm keyway 

[image]

- [Gear 2:](https://www.aliexpress.com/item/4000687927577.html?spm=2114.12010612.8148356.5.5bf26363KskqG9) 1.5M-45T w/10mm bore x 4mm keyway [need to request key cutout])
[image]

_**Stage 2 Gears:**_

- [Gear 1 (same as motor gear):](https://www.aliexpress.com/item/4000896303975.html?spm=2114.12010612.8148356.5.272b39fdOe3OUN) 1.5M-15T w/ 10mm bore x 4mm keyway 

- [Gear 2:](https://www.aliexpress.com/item/4000687938693.html?spm=2114.12010612.8148356.7.5bf263639MZdLT) 1.5M-50T w/ 10mm bore x 4mm keyway [need to request key cutout])
[image]

**Pricing:**
Stage 1 Gear 1: $4.34
Stage 1 Gear 2: $8.18
Stage 2 Gear 1: $4.34
Stage 2 Gear 2: $8.91

kammce, 2020-12-01:
Is the decision to use compound gears then for the shoulder motor?

wilton-lee, 2020-12-01:
Yes

kammce, 2020-12-01:
Wasn't the hierarchy of design complexity something like this (from hardest to easiest):

1. Custom compound gears
2. Custom worm gearbox  
3. Custom pulley gear system
4. Purchase and integrate keyway worm gearbox
5. Purchase and integrate keyway planetary gearbox

Any particular reason why we went all the way back to compound gears?

wilton-lee, 2020-12-01:
1. This design is actually not too complex, would only require one custom
bracket to hold the shaft and gears for the gearbox. I showed David a rough
design and it doesn't seem that complicated. Also allows more flexibility
if we want to increase gear ratio in the future. All of these parts exist
on aliexpress too.
2. This would require us to design a custom shaft since the RMD-X7 shaft is
not long enough to hold the worm, without the motor body interfering with
the shoulder plate. Attaching the gear to the shoulder would also require
an additional shaft since the gear would be too small to drill holes
through for mounting. Complexity wise, I'd say it is about the same as
option 1, but I think David has additional reasons why we preferred option
1 over the custom worm gear.
3. I am assuming this means a simple 10:1 gear ratio without any sort of
gearbox. I looked into small 15teeth/150teeth gear (and pulley) setups and
the 150T gear is extremely large and heavy (plus I couldn't find anything
past 150T) so if we wanted to increase the gear ratio in the future, it
would not be possible without changing the design completely.
4. I thought we didn't want to go with the prebuilt keyway worm gearbox
because they were gigantic and very heavy. And even with this option, we'd
likely still have to design and manufacture additional parts to mount the
RMD-X7 properly to the gearbox. The motor body could potentially also be
too large and interfere with the shoulder plate, but I can't say for sure
without having the gearbox in CAD.
5. I don't recall this option being brought up. I did look at prebuilt
planetary gearboxes but those seem to be designed for specific motors, and
I couldn't find one with a 10mm keyway.

kammce, 2020-12-01:
1. Ok sounds like a decent plan, but I'm not very comfortable with the should being such a custom part as it has to take exceedingly high loads and if we are not careful in how we pick our gears, our gear could shatter. 
2. Acknowledged. so I see where the complication comes from in this.
3. Acknowledged. This pulley system for this ratio is not feasible.
4. Yes, the gearbox could be heavy, but thats just because those were the first gearboxes I could find. I stated in the chat that they were far too large for our purposes and we would have to find something else.  Unless we feel like we wouldn't be able to find a worm gearbox that could fit our needs, we should continue to persue this.
5. This is a new one that I had not brought up yet. I found some light weight planetary gearboxes on AliExpress and though they could work but I didn't look into an exact one that could work. I may just start looking now to see if this is futile or something that we should go for. And I have noticed that many of them are for nema motors, but as long as the shaft fits and doesn't interfere with the body of the RMD-X7 we should be fine.

kammce, 2020-12-01:
Ok, so something concerning I'm noticing as I look for gearboxes is the fact that many of the ones that I find, even the bulky ones, have a maximum rating of 25Nm. Which makes me think that we need to talk to these factories before purchasing these gears to make sure that they can support the 65Nm we plan to use them for.

Specifically many of the planetary gearboxes do not have sufficient maximum loads even though their gear ratios workout (their input shaft dimensions do not work either, but thats not the point)

wilton-lee, 2020-12-01:
Does it specify if the max torque rating is for the input or output shaft?
It makes a little more sense if the input torque limit is 25Nm because with
the 10:1 ratio the output will be 250Nm.

kammce, 2020-12-01:
It doesn't state which it is, but in general, most gearbox motors have similar torque output limits of <45Nm. So seeing these gearboxes state that the load is limited to something like 25Nm makes sense. Our previous motors claimed 5kg-m performance but when we used them with stresses that the arm produces when we pick stuff up, the internal gears shattered. And even though the ratio was correct the size of the internal gears were tiny and not really meant for that kind of stress.

We NEED to ping the manufactures of these parts to get confirmation about how much stress these parts can take. 

I like this series of motors (although the gearbox doesn't have the correct input shaft diameter) and see what they say.

https://www.aliexpress.com/item/1005001708668098.html

wilton-lee, 2020-12-02:
That makes sense for the planetary gearboxes. We may want to go with a worm gearbox just because they are generally rated for higher torques. I found this [NMRV-50 worm gearbox ](https://www.aliexpress.com/item/32556440492.html?spm=a2g0o.productlist.0.0.33fd6312DB6EZn&algo_pvid=b4a8a10a-dc3d-4243-b344-ccf5bcb73bbd&algo_expid=b4a8a10a-dc3d-4243-b344-ccf5bcb73bbd-0&btsid=0bb0624716068684464313872ea740&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)  with our required output torque specified from a [different source](https://www.tecowestinghouse.com/pdf/gr_VSFseries.pdf) (would need to confirm with aliexpress manufacturer)

kammce, 2020-12-02:
I found a [NMRV-30 worm gearbox](https://www.aliexpress.com/item/4000234604090.html) which is smaller, maybe lighter (1.2kg) and seems to fit eh shaft sizes as well. I'll contact them to ask them about their maximum output torque.

wilton-lee, 2020-12-02:
Also I just realized that the 65Nm I calculated was with the arm fully extended and realistically we wouldn't be lifting the 5kg like that. If we controlled the elbow to be hanging straight down, our lever arm would then be only the length of the shoulder link (0.46m) giving us a required torque of 32Nm.

kammce, 2020-12-02:
I think I mentioned that before multiple times, but it totally slipped my mind when we were discussing the arm torque this time around. And yes, it doesn't make sense to pick up an object of 5kg with an outstretched arm. You wouldn't do it with your own body, and it shouldn't be allowable for the Rover. Mission control pilots will have to keep that in mind when they operate the rover which is a perfectly valid and initiative requirement for the arm pilot when they pilot operates the arm.

So if we only need 32Nm of torque, which is the same as the elbow, we only need that same pulley system for this to work, correct? And keep in mind that the arm's motors can pulse torque above 6Nm for a short duration of time. That could help us get a heavy object into a lower torque position (curl the arm in to reduce the lever arm).

wilton-lee, 2020-12-02:
Sounds good. Would it make sense to use gears instead for the shoulder to
avoid potential slippage of the belts?

kammce, 2020-12-02:
> Would it make sense to use gears instead for the shoulder to avoid potential slippage of the belts?

Honestly I'm not sure. I don't have a lot of experience with belts, but I think that our assumption that the belt will slip is a bit overblown. I'm confident that the belt will stay in place.

Also, if you are worried about the belt slipping for the shoulder at the same torque we expect the shoulder to have to deal with, then we would need to move the elbow to gears as well.

wilton-lee, 2020-12-02:
Is there a reason we'd want to use pulleys over gears? The only reason I
can think of is that it is slightly lighter but there are more
possibilities of failure when using a belt vs gears.

kammce, 2020-12-02:
The purpose of pulleys is that you can place the gear far away from the driving gear location. This is a requirement for drive propulsion. It is also a requirement for elbow. Not 100% a requirement for wrist, but the new point of record (POR) for wrist is to use an offset design where the motors are placed away from the wrist in order to bring the weight closer to the shoulder.

So then it becomes a question of, what can actually use gears rather than a pulley? Well, the only thing I can think of is shoulder. But the neat thing about this 32Nm torque, discovery is that the shoulder and elbow can share the same exact hardware, thus reducing the BOM count further. 

Basically it comes down to, do we really need additional hardware for the shoulder when we can just buy more of the same pulleys for shoulder.

Not only that, but we can use the same 15T input pulley gear for drive propulsion as well. Maybe we use a different output pulley gear since we only need 14Nm of torque for the rover as calculated by #73. (FYI: @HaadiElahi @davidbdias96) If we used the 80T pulley we would get ~20RPM of motion (5:333:1 ratio with a top speed of 105RPM), which is really slow. That would mean the top velocity for the rover is 0.34578544 m/s or better yet, it will take 3 seconds to move 1 meter.

wilton-lee, 2020-12-02:
That's fair, then the input pulley would be the same for elbow, shoulder, and drive propulsion and then the output pulley would change based on our desired torque.

kammce, 2020-12-02:
Correct. And the output pulley would be the same for elbow, shoulder. The output pulley would be different for drive propulsion. 

I feel like this is nearly ready for closing. @wilton-lee can you do the following things:

1. Make a concluding comment on this issue with just the end solution to the problem.
2. Update the first comment to indicate that the stuff you put there is not the final solution
3. Update the first comment with a link to the concluding comment stating that this is the approved solution.

wilton-lee, 2020-12-02:
**Approved Solution: belt system with 5:1 gear ratio (same as elbow)**

- Required Torque: 32 Nm with optimized shoulder/elbow control 
- Gear Ratio: 5 to 1
- Input Pulley: [XL-15T](https://www.aliexpress.com/item/4000389405414.html?spm=2114.12010612.8148356.3.76165349Nlo0UH) (11mm width, 10mm bore, 4mm keyway)
[image]

- Output Pulley: [XL-80T](https://www.aliexpress.com/item/32354211180.html?spm=a2g0o.productlist.0.0.367b5194RuV9eB&algo_pvid=1dd37e69-1ba8-4297-87e9-3f12166b92e0&algo_expid=1dd37e69-1ba8-4297-87e9-3f12166b92e0-4&btsid=0b0a556316069281297444392e9cdc&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) (11mm width, bore N/A)
[image]

closed 2020-12-02 by wilton-lee