SJSURoboticsTeam/urc-central-2021 #141: Elbow Gear Ratio/Design

wilton-lee, 2020-12-01:
**ELBOW**
Required Torque: 32 N-m
Desired Gear Ratio: 5:1

- [Pulley 1 - Motor Pulley:](https://www.aliexpress.com/item/4000389405414.html?spm=2114.12010612.8148356.3.76165349Nlo0UH) XL-15T w/ 11mm belt width x 10mm bore x 4mm keyway
- [Pulley 2 - Driven Pulley:](https://www.aliexpress.com/item/4001040586841.html?spm=a2g0o.productlist.0.0.3d151935ySRaf4&algo_pvid=9e40a0fe-55af-4aba-8f4c-1ba530ffc3ed&algo_expid=9e40a0fe-55af-4aba-8f4c-1ba530ffc3ed-6&btsid=0b0a555b16067943419807634e16d3&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) XL-80T w/ 11mm belt width 

**Pricing:** 
Pulley 1: $3.37
Pulley 2: $21.17

kammce, 2020-12-01:
Looks good to me. I'm wondering what we can do to mount the driven pulley to the arm's elbow? Do we do something like drill 4 holes into it so we can screw it into the elbow?

wilton-lee, 2020-12-01:
Yeah that was our plan, which is why I wasnt worried about the bore hole
for the larger pulley.

kammce, 2020-12-01:
Awesome, sounds good to me then.

wilton-lee, 2020-12-02:
With the reduced torque requirement if we lift the 5kg at an optimal trajectory, we could theoretically use a smaller pulley on the elbow too. Having an 80T gear on the elbow is kinda bulky if we don't need that gear ratio.
[image]

kammce, 2020-12-03:
> With the reduced torque requirement if we lift the 5kg at an optimal trajectory, we could theoretically use a smaller pulley on the elbow too. Having an 80T gear on the elbow is kinda bulky if we don't need that gear ratio.
> [image]

True, but then we would add another component to our BOM count. I think it is totally responsible and correct to reduce the gear size for elbow in general because we don't need the added torque. But, if the design looks a bit clunky you can UN-clunk-it aesthetic wise by simply making the elbow joint metal area as wide as the gear. Then you have it taper down to the wrist required lengths as it is now. You could do a similiar thing with the shoulder too, but I don't think it looks too bad. Widening up the elbow area may be required anyway because the wrist RMD-X7 motors are going to be moved closer to the elbow area and drive the diff gearbox using pulleys, to bring the CG back towards the base of the arm.