SJSURoboticsTeam/urc-central-2021 #143: Gimbal - Design and Manufacture Mount for Camera

maximvoven, 2020-12-02:
Design and manufacture mount for a rpi camera to bolt to gimbal that is capable of being counter waited to properly balance the gimbal with minimum weight.
[rpi Camera Mechanical Specifications](https://www.raspberrypi.org/documentation/hardware/camera/mechanical/rpi_MECH_Camera2_2p1.pdf)
The mount should allow for the placement of weights along the top and right side of the mount, these weights will be used to bring the gimbal into equilibrium so that external accelerations did not impose significant torque on motors.
The mount will likely be 3d printed.
The raspberry pi will not be mounted on the gimbal instead the flex cable will out the right side of the mount and reach to rpi mounted on mast. The raspberry pi will be hooked up to gimbal controller in the hope that pitch control can be established in the future.

kammce, 2020-12-03:
Can you format the first comment using markdown to help separate the list of bullet point requirements so its easier to read. Right now its kinda a blob of text.

kammce, 2020-12-03:
Is the idea that the RPI Zero camera would be the device that controls the mast gimbal? 

I think the original idea for this was to have the Arm board control the mast positions as this should be rather simple. 

## Here are some attributes for this decision

- Code in Arm is already prepared and ready for communication with mission control, so adding more commands is little effort
- With the introduction of the RMD-X7 motors, only CANBUS and I2C, for the IMUs, are utilized, meaning we have a bunch of available pins for controlling pitch and roll.
- Pitch and roll tend to be controlled using RC servo signals and do not require monitoring, meaning, we just set the PWM and leave it be.

## Some of the CONS of the arm board approach

- You would need to wire GND and PWM signals (single-ended signal) all the way from the Arm board up to Mast which is a bit far away, potentially up to 1 meter away.
    - A counter to this though is that the signal integrity should be fine at this distance. Personally, I've seen PWM signals go from a small lab of 10 meters and end up on the other side without much degradation.

## Some of benefits of using the RPI Zero

- Isolate mast control from Arm Controller
- Makes more logical sense to have the mast camera control the mast

## Some CONS for the RPI Zero

- Would require the team to learn how to write software for the RPI Zero which utilizes a different language, and adds to the set control system design complexity. Where we originally had just 1 language, 1 API and two boards to control, now we have 2 languages, 2 APIs and 3 boards to control.
    - Counter to this is that  can be done with python and is overall pretty easy to learn and work with.