SJSURoboticsTeam/urc-central-2021 #144: IMU Adress Clash Issue

maximvoven, 2020-12-02:
We have 4 imu's that need to be mounted on arm for homing, issue is that current imu we have selected for use has only 2 possible addresses. 
Electrical is currently exploring two possible actions:
1. Use two brands of imu's with different addresses.
2. Address translate two of the imus.

kammce, 2020-12-03:
You may want to throw one more option into the fold here. 

3. Use two I2C buses

The Stm32f103CU MCU that we are using has two I2C buses, so for the reference IMU and shoulder IMU can be on I2C1 and elbow and wrist IMU can be on I2C2.

maximvoven, 2020-12-06:
Noted, I have requested samples of LTC4316 i2c address translators, Ill build a prototype board and we will test to see if we can use it. If we run into issues with signal integrity and to many i2c errors we will only then order two different imu's. This is being done to avoid rewriting new drivers.

kammce, 2020-12-06:
> Noted, I have requested samples of LTC4316 i2c address translators, Ill build a prototype board and we will test to see if we can use it. If we run into issues with signal integrity and to many i2c errors we will only then order two different imu's. 

This is good. It will be good to see if we can make these devices work properly as before, it had not been achieved. I'm certain it will work and that mistakes were made previously that prevented the translators from working.

> This is being done to avoid rewriting new drivers.

I'm not sure what you mean by this. If we have an I2C peripheral driver, that driver is reusable across all of the I2C ports in general for almost all MCUs. Some devices will have slightly different register layouts to promote additional special features, but if we get 1 I2C bus working, then we have them all working.

Also, if you are looking to simplify the complexity, omitting the LTC translators would be the way to go, as we would simply need to keep the two sets of IMUs on different buses and call it a day.

And to keep perspective, the rover needs to be done somewhere in early January in order to give controls and the team a chance to make the rover work enough and tested enough to actually be able to produce a video report for competition. So if we can reduce complexity for the first iteration (we can always be fancy in the next iteration) then we should.

maximvoven, 2020-12-22:
Update: Received Sample Chips waiting for breakout board for chip package to build prototype units.

kammce, 2020-12-22:
@maximvoven may have already come up and voiced this idea prior, but I thought it bares the need to be written here.

Something I had considered a week ago regarding the IMU address clash. I was thinking back to the usage of USB-C on the Rover and I think the usage of the cable should be limited to just the RMD-X7 motors. Potentially just for Drive and the Wrist motors as they are far away from their control boards. The shoulder and elbow motors on the arm board will probably be so close to the board that a direct ZH1.5-8 connector can be used directly. Using the ZH1.5-8 connector will not increase the BOM lines as it will be used anyway for the final connection from USB-C to the motor via a breakout board.

Something to note about the USB-C 2.0 cables I've worked with is that they DO NOT include the SBUx lines, just VCC, GND, D+, D- and CCx, so the pin count is a bit more limited than expected.

My suggestion for the IMU connector is something like this:

| RJ45 Pinout | Signals |
|-------------|---------|
| PIN1-GW     | SCL     |
| PIN2-G      | VCC     |
| PIN3-OW     | CS-IMU1 |
| PIN4-B      | CS-IMU2 |
| PIN5-BW     | CS-IMU3 |
| PIN6-O      | CS-IMU4 |
| PIN7-BrW    | GND     |
| PIN8-Br     | SDA     |

The idea here is to daisy chain ethernet connectors from the shoulder to elbow to wrist. And since each node in the chain will have access to all 8 pins of the ethernet connector, we can specialize specific pins for specific devices. In this case, power and communication take up 4 pins and the last pins are chip select pins that connect to the AD0 pin of the MPU6050. This would eliminate the need for additional ICs such as the LTC4316 and makes each daisy chain board containing the MPU6050 as simple as two ethernet adapters, a decoupling capacitor, a place to slot in the MPU6050s and a dip switch or shorting block used to select which CS changes the Address pin for that node. 

Again I still believe that it is extremely useful to get the LTC4316 chip working so that we know we can get that chip working and figure out any caveats with the IC in the event we need it in the future.

maximvoven, 2021-06-03:
LTC4316 has been forced into working, issue is non issue

closed 2021-06-03 by maximvoven