SJSURoboticsTeam/urc-central-2021 #152: Parse Mission Control JSON Data

naterpotatoers, 2021-01-26:
Implement arm and drive system JSON parser for incoming mission control data. 
Example JSON Data (string):
`{ "is_operational": true, "drive_mode": "T", "speed": 80, "angle": 45} `

Extract the value such that we know
is_operational = true
drive_mode = T
speed = 80
angle = 45
Should be possible using [sscanf](https://www.youtube.com/watch?v=-7cSmcdMryo)

kammce, 2021-01-26:
Gonna spoil this for yah,

the algorithm for this is kinda a one liner:

```C++
int parsed_elements = sscanf(server_data, "{ \"is_operational\": %d, \"drive_mode\": \"%c\", \"speed\": %d, \"angle\": %d}", &is_operational, &drive_mode, &speed, &angle);
```

Then you check the `parsed_elements` to make sure the number is correct. If it isn't, then the values may either be not updated or they were not updated correctly.

naterpotatoers, 2021-02-05:
created PR

kammce, 2021-02-05:
When you create a PR make sure to reference these issues in their commits. 
For example:

```
Resolves SJSURoboticsTeam/urc-central-2021#152
```
which turns into this when done properly:

Resolves SJSURoboticsTeam/urc-central-2021#152

The PR will pop up on this issue and anyone looking at this issue will know why and what closed it. Rather than us just closing them without indicating to anyone following along why we did so.

closed 2021-07-09 by naterpotatoers