SJSURoboticsTeam/urc-central-2021 #153: Implement linear interploation handler for drive systems

naterpotatoers, 2021-01-27:
Implement linear interploation (exponentional moving average) to smooth out changes in speed.
Refer to SetWheelSpeed() within rover_drive_system.hpp

naterpotatoers, 2021-02-04:
std::lerp might be a solution but this would require cmath which might be too cumbersome

kammce, 2021-02-04:
I'm not sure why including <cmath> would be cumbersome. It would be far less cumbersome then you implementing std::lerp yourself. Lets not reinvent the wheel when we don't need to. 

But just using lerp won't solve the problem. You need to lerp at several time intervals to produce what looks like acceleration.

naterpotatoers, 2021-02-05:
Noted. I was thinking that maybe cmath would be too large of a package for the microcontrollers. Do you think that using a fixed delay between lerps would be good solution? i.e.
```c++
lerp(current_speed, target_speed, 0.25f);
sjsu::delay(500ms);
lerp(current_speed, target_speed, .50f);
sjsu::delay(500ms);
... --> .75f --> 1.0f
```
Or what would you suggest as a better time interval?

kammce, 2021-02-05:
With lerp you set a fixed value for alpha. You don't change it. Also you don't want to have multiple delays in your code as that will mean that, for 500ms you are not processing anything. It would look something more like this:

```C++
auto new_speed = std::lerp(previous_speed, target_speed, 0.5f);
SetMotorSpeed(new_speed);
```

Now you may ask where the delay comes from? Well you have a delay between each call t mission control. In your single threaded code (and in multi-threaded code as well) you want to ensure that there is just a singular delay that controls how fast things operate. BUT! You may notice that if we wait 100ms per lerp step that would probably be super long, so we may need to speed the main loop sample rate something like this:

```C++
int loop_counter = 0;
for (int loop_counter = 0; ;loop_counter++)
{
  if((loop_counter % 10) == 0)
  {
     GetMissionControlData();
  }

  OperateDriveSystems();
  sjsu::Delay(10ms);
}
```

By changing the delay to 10ms we can lerp the motor speed 100 times a second, creating smoother control of the speed. But we don't want to get data from mission control that fast. We only need it every 100ms, so we can use a loop counter and check if it is divisible by 10. Everytime the counter counts up to 10, the modulus is zero and we can get our latest data that way.

This is usually how you would handle single threaded systems. One single loop and a bunch of systems that check if its their turn to operate. All of them must ensure that none of them delay, otherwise the timing would be off.

Going 1 step even further, to ensure that the delay time is ALWAYS 100ms, there are a number of options. The issue with using sjsu::Delay() is that the sample rate will always be greater than 100ms if we did it this way. Getting data from mission control, doing calculations like lerp and commanding the motors takes time. Lets say 50ms on average. Thus an additional delay of 100ms will result in a sample rate of 150ms. To fix this, rather than delay for 100ms, you can record the previous uptime count, calculate when the next sample rate time should be and run when the uptime() function is at or above that number.

For example lets say we call Uptime() and we get `1025ms`. Add 100ms to that and we get `1125ms`. The code before the delay takes 50ms so now uptime is `1075ms`. We need to delay/loop for another 50ms before we can run our code again. So that is what we do.

```
while (1)
{
  auto next_same_time = Uptime() + 100ms;

  // Do all of the work you'd normally do at this stage

  while (Uptime() < next_sample_time)
  {
      continue;
   }
}
```

But there is a way that requires less math that you can do using the `sjsu::TimeoutTimer` class

```
while (1)
{
   sjsu::TimeoutTimer timeout_timer(100ms);
   // Do stuff ...

   // If doing stuff took 75ms, then timeout_timer.TimeLeft() will report 25ms
   sjsu::Delay(timeout_timer.TimeLeft());
}
```

`sjsu::TimeoutTimer` gets what the current time is on construction of the object. And each call to TimeLeft() will give you the time between construction the timeout time and the current time.

kammce, 2021-02-05:
NGL though, this may all be moot if we can set the acceleration of the RMD-X motors (their PID P value) and it not cause them to be unstable. The P value can be considered the smoothing or acceleration term. Setting it badly could result in a unstable system that oscillates. But this may not be the case and the motor could potentially do some smart stuff to prevent that. If we can control the P value and it still retains its original behavior, strength and stability, then we can let the motors lerp themselves internally. 

But if they cannot, then we must lerp them ourselves.

kammce, 2021-02-05:
[image]
There is an acceleration parameter, but...

[image]
This states that it doesn't actually work and that we need to hit the PI parameters. Which, if set badly could cause the motors to perform badly. But it should be dangerous to the motor though.

naterpotatoers, 2021-10-13:
Closing this since there is a newer issue #171 that is more current. This is referenced in that issue so this info is documented