SJSURoboticsTeam/urc-central-2021 #163: Research into the i2c converter

naterpotatoers, 2021-03-01:
Don't have one selected yet but might want to look into how it works from a control systems perspective. Not sure how these work data wise.

[Possible converter](https://www.adafruit.com/product/815)

kammce, 2021-03-03:
Renamed the title of this to something that makes more sense when read. I'm not sure how one would interpret i2c converter without knowing what you are trying to convert it into.

naterpotatoers, 2021-03-08:
Looked into the [PCA9685 ](https://www.adafruit.com/product/815)and it looks really nice, perfect for our needs. I think we should definitely move forward with this module. Helpful [example ](https://youtu.be/kUHmYKWwuWs?t=1523)of someone using this specific board with Arduino. Will have Andrew and Ulysses pick them up after meeting up with Electrical to make sure it's okay

kammce, 2021-03-09:
Before closing this out, can you explain in detail why its a perfect fit for the needs of the team?

naterpotatoers, 2021-03-09:
Actually after discussing it further during leads meeting I think this might not be appropriate for our needs. The linear actuators require 7v to power them where as the PCA9685 can only drive up to 6v. Will need to find a better alternative or come up with a clean way to provide power for the linear actuators

kammce, 2021-03-09:
The PCA9685 isn't a power source. Its a PWM generator. Supply the battery voltage to the claw and step down the voltage to the appropriate 7V at the hand of the claw. You shouldn't need to rely on the circuits on the breakout boards for the PCA9685.

naterpotatoers, 2021-03-09:
Yeah that was a minor confusion at my end, not an electrical guy. After clearing it up with maxim he assured me that the actuators are 6v not 7v so my concern is a non issue. That the PCA9685 will be perfect for our needs. @maximvoven if you can maybe explain why the PCA9685 is perfect fit for the needs of the team that would be great so I can close this issue out. Thanks

maximvoven, 2021-03-09:
The board is good because it has a wide selectable i2c address range (lowers odds of address clash), Reasonably priced and specifically designed to drive servos not led's. It also has a trace for providing power, so it can act as the pwm signal generator and power distribution (again because specialized for servos). While 16 channels is a little excessive it's pretty difficult to find good boards with less channels that keeps the address flexibility of i2c address selection of this board and being specifically designed for driving servos.

kammce, 2021-03-09:
Sounds good to me. 👍🏾

closed 2021-03-10 by kammce