SJSURoboticsTeam/urc-central-2021 #183: Document Work for Arm System

naterpotatoers, 2021-10-13:
Might be good idea to do this in coordination with new controls members. Basically describe all the functionality the Arm needs to achieve to give basic outline for drafting Arm code. May want to do this with #182 first since they are already familiar with that.

naterpotatoers, 2021-11-20:
A little google doc @Coreyboy1820 made [here](https://docs.google.com/document/d/1Eyw-78VcC1dD00w_sQ50Rw7b76E4pKceh4LsTbSEcUM/edit)

Coreyboy1820, 2021-11-23:
To bring up the topic on the organization of the inputs from the server, do we need information from the sensor sent to each motor or one value and all the motors operate under this single value? I understand we need: 1. Speed of arm 2. Angle of motor (multiple or one) 3. Values for hand motors (since these should be different from arm motor values). Depending on how we want to tackle the implementation of the arm.... We can give the hand modes, values, or even both. For example, open and closed mode or values for each motor in the hand. Another implementation issue we need to talk about is the single value for the motors or multiple values for the motors, we can pass in one value to one motor (the lead motor of the hand) for example the middle finger or thumb and have the other 4 fingers follow in a similar fashion when closing. What do you guys think? We are in the drawing board stage i'm assuming.

josh-github1, 2021-12-04:
This ties into https://github.com/SJSURoboticsTeam/urc-central-2021/issues/184.. @kammce mentioned we need to use the accelerometer for homing.. not sure how reliable the accelerometer data already is or needs to be but if the homing is inaccurate then the angles we want to set our motors/joints to on the robot arm could subsequently be inaccurate..

Maybe a few things we could do with the mpu6050.. it has gyroscope data and we can combine that with the accelerometer data.. I remember people used accelerometer data for their trial project but maybe not gyroscope data. I think the current mpu6050 driver does not output gyroscope data but I could totally be reading the code wrong - https://github.com/SJSU-Dev2/SJSU-Dev2/blob/master/library/devices/sensors/movement/accelerometer/mpu6050.hpp

We could also do something fancy like a low-pass filter 0 https://www.pieter-jan.com/node/11

josh-github1, 2021-12-04:
idk people's familiarity with gyroscope but it basically gives angular velocity along 3 axes: https://learn.sparkfun.com/tutorials/gyroscope/all

kammce, 2021-12-04:
It doesn't need to be very accurate actually. We just need it within a 60 degree band which is pretty massive which has to do with how the motor does position control.

josh-github1, 2021-12-04:
Ah sounds good! Thank you for letting me know!

josh-github1, 2021-12-04:
tbh I'd personally like to explore these purely out of interest but probably not the best use of my time to help finish objectives for the robotics mission. Will probably look into it later.

closed 2022-01-19 by naterpotatoers