SJSURoboticsTeam/urc-central-2021 #190: Implement Speed Logic for Drive Mode

naterpotatoers, 2021-12-29:


josh-github1, 2022-01-21:
@naterpotatoers Can you elaborate on this? I'm interested in what you mean by speed logic

josh-github1, 2022-01-21:
Thank you Nate!

josh-github1, 2022-02-16:
@naterpotatoers Does drive mode mean that no wheels are turning? Also, has anything like this already been implemented that we can reference?

I'll try to review https://github.com/SJSURoboticsTeam/urc-control_systems-2019/blob/Drive/Projects/Drive/main/Motor_Control_rev1.cpp

josh-github1, 2022-02-16:
I found a paper that uses the third wheel for steering with the 2 other wheels maintaining a constant angle

https://arxiv.org/pdf/1612.01476.pdf

josh-github1, 2022-03-23:
Also one difference is that this paper does PID control and we're doing the Lerp method.. I think it's good either way

closed 2022-04-27 by naterpotatoers