SJSURoboticsTeam/urc-central-2021 #217: Remove float type from MissionControlData structure

naterpotatoers, 2022-01-13:
MissionControlData should use int type since that is the only type that is sent to us from the web server. This would remove all the warnings that are appearing in Arm. This would also allow the data types to match mission control correctly since it will only return int values.

All the calculations may need to modified slightly as a result since we had been running calculations directly on the mission control data.

Coreyboy1820, 2022-01-14:
Worked on it, however I don't see where any calculations would have to be modified, if someone sees something please fix it for me or leave a example below for me to understand, thanks.

naterpotatoers, 2022-01-14:
I guess theres no calculations but for a few functions we pass integers to functions which require floats which causes the following warnings.
`/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:158:32: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  158 |     rotunda_.SetSpeed(mc_data_.arm_speed);
      |                       ~~~~~~~~~^~~~~~~~~
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp: In member function 'void sjsu::arm::RoverArmSystem::MoveShoulder(float)':
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:164:33: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  164 |     shoulder_.SetSpeed(mc_data_.arm_speed);
      |                        ~~~~~~~~~^~~~~~~~~
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp: In member function 'void sjsu::arm::RoverArmSystem::MoveElbow(float)':
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:170:30: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  170 |     elbow_.SetSpeed(mc_data_.arm_speed);
      |                     ~~~~~~~~~^~~~~~~~~
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp: In member function 'virtual void sjsu::arm::RoverArmSystem::HandleRoverMovement()':
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:177:26: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  177 |     MoveRotunda(mc_data_.rotunda_angle);
      |                 ~~~~~~~~~^~~~~~~~~~~~~
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:178:27: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  178 |     MoveShoulder(mc_data_.shoulder_angle);
      |                  ~~~~~~~~~^~~~~~~~~~~~~~
/home/naters/code/urc-control_systems-2021/Arm/rover_arm_system.hpp:179:24: warning: conversion from 'int' to 'float' may change value [-Wconversion]
  179 |     MoveElbow(mc_data_.elbow_angle);`

closed 2022-01-15 by naterpotatoers