SJSURoboticsTeam/urc-central-2021 #222: Manually mock mpu6050

naterpotatoers, 2022-01-17:
Until we figure out why its not working lets just create a mocked mpu6050 manually so that way we can continue using the mpu as arguments for the various things used in arm

acarrou, 2022-01-24:
I got the MPU to work using the 0x34 address instead of 0x68. After talking a little with Khalil, he suggested the problem could be that the boards I was using were know-offs due to the chip shortage. The code works when changing the expected device address in the MPU class to 0x34.

kammce, 2022-01-24:
Oh, also, you can just mock the `sjsu::Accelerometer` class and just use that. You don't need the MPU class mocked directly. Just its interface. All accelerometers basically act the same so you don't need to mock MPU specifically.

acarrou, 2022-01-24:
> Oh, also, you can just mock the `sjsu::Accelerometer` class and just use that. You don't need the MPU class mocked directly. Just its interface. All accelerometers basically act the same so you don't need to mock MPU specifically.

Ahh okay, I'll mock that up with @ryerlic thank you for mentioning that.

closed 2022-02-19 by naterpotatoers