SJSURoboticsTeam/urc-central-2021 #227: Develop a driver for the i2c to PWM 16-channel device

josh-github1, 2022-01-21:
Thank you @naterpotatoers for the reference: https://github.com/SJSURoboticsTeam/urc-control_systems-2021/blob/master/Arm/Hand/pca9685.hpp and credits to @AndrewDEmerson for working on this.. I think this could be .hpp file in the Common folder or maybe even in the SJSU-Dev2 code?

kammce, 2022-01-22:
The title of this issue is to make an driver for the pca9685 (the i2c to 16 channel pwm device). But the first comment references an already existing implementation. What is the purpose of this bug?

josh-github1, 2022-01-22:
Hey Khalil, whoops! I probably made a mistake or there was miscommunication.. I thought this still needed to be fully implemented/tested but if it's already finished then we could close this one out.

kammce, 2022-01-23:
You should list in the first comment exactly what needs to be done so its clear what is needed. Also change the title to reflect the intent of this issue.

josh-github1, 2022-01-23:
Thank you for the tips! I haven't looked into the code fully but wanted to create this issue if it isn't fully implemented. I'll make the changes soon unless anyone wants to edit it. I didn't assign myself yet

closed 2022-04-27 by naterpotatoers