SJSURoboticsTeam/urc-central-2021 #229: Implement pca9685 pin values into finger constructor

naterpotatoers, 2022-01-26:
After meeting with Andrew we got better idea of how to our robots finger logic. Now we need to create the logic for storing the PCA96855 pins for the fingers in a local variable probably in finger.hpp e.g.  

- thumb = 0
- index = 1
- middle = 2
- ring = 3
- pinky = 4

Then we should probably think about how the hand class will call upon the fingers using the PCA9685 module to move properly.

Coreyboy1820, 2022-02-09:
So there's no specific angle to set the fingers, its just stationary = 0, 1ms = retract, and 2ms = extend? Or will our driver handle and convert it to angles?

Andrew-Ingerman, 2022-02-09:
There is supposed to be a function that converts it to angles

Coreyboy1820, 2022-02-09:
Alright thanks. Are you working on this right now?

Andrew-Ingerman, 2022-02-12:
> Alright thanks. Are you working on this right now?

Yeah

Andrew-Ingerman, 2022-02-12:
Move Finger logic back into hand so we don't replicate the PCA 5 times.

Coreyboy1820, 2022-02-14:
> * Move Finger logic back into hand so we don't replicate the PCA 5 times.

Can you tell me why we need to do this rather then putting the PCA in the hand class itself?

Coreyboy1820, 2022-02-14:
Doing this would make us need to create 5x the code of the finger class inside of hand.

naterpotatoers, 2022-02-17:
Yeah but it removes the complication of have 5 instances of the pca device. We might go back to finger class once we know that it works with this much simpler solution

closed 2022-04-27 by naterpotatoers