SJSURoboticsTeam/urc-central-2021 #234: Update mpu member variables to use accelerometer

naterpotatoers, 2022-01-29:


kammce, 2022-01-31:
What, what does this mean? Are you trying to change the MPU class itself or doing something else?

naterpotatoers, 2022-01-31:
this is to resolve the tight coupling issue with the mpus. Switching the member variables to be Accelerometer object as opposed to mpu6050

kammce, 2022-02-01:
Ah ok. The wording of the title is very confusing. You may want to change this to be more direct such as "Change references to Mpu6050 to sjsu::Accelerometer".

closed 2022-02-11 by Coreyboy1820

kammce, 2022-02-11:
Why was this closed? Was this linked to a PR? If so list it here or make sure that you have `Resolves #234` in your commit message so it automatically closes issues when the PR is merged.

Coreyboy1820, 2022-02-11:
Thanks @kammce I didn't know that was a thing. I'll do this next time. I was just trying to take care of issues that have been resolved for a few days.