SJSURoboticsTeam/urc-central-2021 #237: Move Acceleration struct into its own file

naterpotatoers, 2022-01-29:
I've noticed we keep recreating this struct since they its usually defined within a class themselves. This seems a bit silly since we have to then recreate the struct if we want to use it in another class.

By making this struct be fully independent of a class in their own files we can more easily update and manage the values in one centralized location reducing duplication. Not sure if this is a good practice but it makes sense to me moving forward

naterpotatoers, 2022-01-29:
Also this would clean up the unit tests since we have to redefine the structs there too usually... Maybe we could move all structs within their own files for clarity

Coreyboy1820, 2022-02-04:
I don't see why not. However I just fixed this issue, I thought I removed all instances of acceleration other than in the joint class.

naterpotatoers, 2022-02-19:
okay alternative solution was found

closed 2022-02-19 by naterpotatoers