SJSURoboticsTeam/urc-central-2021 #242: Add const keyword to accessor functions

naterpotatoers, 2022-02-11:
In Data Structures and Algorithm Analysis in C++ by Mark Allen Weiss he claims that accessor (getters) should be marked with the `const` keyword as it's very important part of the design process. Otherwise by default all member functions are technically mutators by default. This is technically true for any locally scoped variable that shouldn't have its value updated as a best practice to prevent possible bugs

acarrou, 2022-02-11:
> In Data Structures and Algorithm Analysis in C++ by Mark Allen Weiss he claims that accessor (getters) should be marked with the `const` keyword. Otherwise member functions are technically mutators by default. This is the same case for any locally scoped variables that shouldn't have its value updated as a best practice to prevent possible bugs

ah interesting, never thought of this but sounds like it's a good way to avoid bugs!

Coreyboy1820, 2022-02-11:
Looking into this now.

acarrou, 2022-02-11:
> Done for arm, just have to get someone to do it for drive.

Maybe we also do it for the rover_arm_system.hpp file?

naterpotatoers, 2022-03-06:
done

closed 2022-03-06 by naterpotatoers